wolkstein/MavLink_FrSkySPort #1: Wrong GPS coordinates are shown on standart telemetry screen

Linktuner, 2015-01-11:
While using converter based on arduino nano 328p with no functional change in code of teensy there are wrong coordinates on the standart telemetry screen. So it has no chanse to use it for the flown away model search.
[image]
it sould be like about 56.312283, 38.146166

Davestru, 2015-01-11:
We found a bug in the code.
First, you need to do:
(int32_t)(ap_latitude / 100 * 6);
then compare it to zero, and after multiplied by a hexadecimal number.
We collected his version of the image with the correct GPS, but there is a suspicion that it is possible to work properly radar.
Please provide correct code and check the radar!

wolkstein, 2015-01-21:
thx for pointing out. i never compare this coordinates with real coordinates because the radar works relative to home position. so i will have a look on it.

wolkstein, 2015-01-28:
i do some tests today and can not confirm this bug. at my taranis, lat & long are shown correctly 
env:
Taranis with opentx 2.0.15
x8r receiver
teensy 3.1 with resistor network to measure lipo cells.
apm 2.6 

this bug need more data.

@Linktuner, you use an arduino nano. this repository contains arduino firmware code to compile and run at an teensy3.1. it is not possible to compile this code for an nano without rewriting at minimum the serial setup. also 

@Davestru, against which version of GCS_MAVLink did you compile the code. did you use an teensy3.x? and which version of opentx didyou use.

wolkstein, 2015-03-28:
fixed in https://github.com/wolkstein/MavLink_FrSkySPort/tree/s-c-l-v-rc/MavLink_FrSkySPort_Teensy2_and_ProMini folder.
see
6b3988367931f52359385530baec6e517caf331a
and
6213e7d40a5dc92b2415cf7f254699f609627fd8
/g
wolke

closed 2015-03-28 by wolkstein

FPVCambodia, 2015-03-30:
Hi Wolke,

Tks again for your help (on DIY Drones), here is my small bird alive with
your help and readme tutorial on GitHub. I now need to calibrate the Lua
scripts after a few packs as you suggest... and I should be good with a
redundant info on my Taranis.. This small Trifecta Tricopter bird, features
a GPS, osd, Mavlink 3DR, Mavlink to FrSky ( ;) ) and low KV motors with a
4000mAh 3S battery set for 10 to 12 minutes flight...

Here are a few pictures for your reassurance that the community goes to the
end making it working and helpful supporting us forward! Thanks again to
you and Rolph! Well done guys and Tks Again!

https://plus.google.com/+FPVcambodia/posts/M9Abt32Htvq

Kind Regards