wolkstein/MavLink_FrSkySPort #4: Telemetry not displaying

kd5ftn, 2015-04-20:
On my Taranis, I'm only receiving audio alerts from the MAVLink data, such as prearm warnings, and flight mode changes. Voltage, current, heading, GPS and other telemetry do not display on the screen - instead I see all values zeroed out. 

Since the red light on the Teensy is flashing quickly, and I'm at least getting audio alerts, I know that data is being transmitted.

I purchased the pre-wired Teensy from Peter King at Droneshop.biz, but have since loaded the LUA / SD Card files and pushed your latest compiled hex to my Teensy, and experience the same issue.

I'm using a Taranis Plus running OpenTX-Taranis-Plus-2.0.15, with LUA enabled. My flight controller is a Pixhawk clone (HKPilot32), and my reciever is an Frsky X8R.

kd5ftn, 2015-04-20:
Edit: Just switched my Teensy from Telem2 to Telem1 and now I get full telemetry data. Off topic for your code, but do you happen to know why that is?

CodeChief, 2015-04-20:
The start of the Teensy code (any any other MAVLink code such as MinimOSD) sends subscription messages and following-on only receives those messages. I don't know if Pixhawk is hard-coded to always output everything on Telem1, but it's possible because it's the suggested port for radio communications.

The problem with using Telem1 is according to the Pixhawk documentation it has also has a higher output to support the 3DR Radio, so I didn't want to use that because I prefer to ensure I have maximum range (somebody with more electronics skills/Pixhawk knowledge please correct me here). Also more detail on the maximum ratings for both ports would be nice, didn't find that on the official pages.

Anyway so I edited the code on my Teensy to get it working on Telem2 (added some more subscriptions, like you see in the MinimOSD code) and also experimented with a throughput of the MAVLink messages back out another serial port of the Teensy so the MinimOSD could be daisy chained. It works!

Also I found there is a bug in the code, the documentation of the main Teensy function suggests the light should only flash when a heartbeat is received (obviously the original design) but later on in the code it is flashed for every single data transfer. I find the original idea better, because the light is only on when it is really considered active.

I'll be writing a blog article about that soon to share the code/configs but to start with I can say it should work on Telem2 with HKPilot32 (which I also have) and Teensy 3.1 although maybe you need my fixes. Similarly if your OSD works on Telem2 then the Teensy should, so are they both broken for you? Sorry I can't give the complete solution right now I'm very busy until later in the week/weekend but will try to get this shared.

kd5ftn, 2015-04-20:
Thanks Tony, 

I haven't tested with an OSD yet, as I'm only using a 3DR radio and the Teensy. From what I've read, both Telem1 and Telem2 provide enough current to power the 100mW 3DR radios, but higher powered radios, such as the 1W RFD900 require more power which only Telem1 can provide.