wolkstein/MavLink_FrSkySPort #5: Amps (battery current) showing zero

athertop, 2015-05-25:
Running latest teensy 3.1 code (downloaded May 24th) with voltage divider for cell values connected to 3s LiPo valance cable.  Amps (Current) values are sitting at zero on the LUA page. All other telem values seem fine aside from Amps (and Watts, plus Wh, which I'm guessing need the Amps value to be calculated). Under my Taranis Model config, Telemetry page, I have my voltage source and current source set at 'FAS'. Teensy 3.1 is connected with a y cable so both it and OSD share the telem 1 port. Only teensy has Tx and rx connected  (OSD only has only its rx port connected to telem 1 tx). I can see amps displayed in OSD, but showing zero in lua screen.   Any help/advice would be greatly received. Thanks, Paul

wolkstein, 2015-06-23:
hi,
sorry because the delay.
did you get current values normally via mavlink? if not maybe you do not set-up apm/pixhawk correctly in missionplanner. normally if you compile this teensy code with default settings (with single cell monitor), voltage is measured via teensy and current is reported from mavlink.

/g
wolke

athertop, 2015-06-23:
Wolke, Sorry, I forgot I posted this. Its been resolved now. It was due to the SRn_ parameters that your teensy code uses. I think it uses 3 mavlink channels/streams in all - I needed to set the values to 10 to get all the relevant telemetry to the teensy. I'm sure I reported this against Rolf's post on diydrones also. Cheers for getting back though.

closed 2016-10-20 by athertop